KRALYN/PositronV3 #1: Magnetic pogo pins

jmsether, 2022-05-21:
[image]
I am having a hard time locating the magnetic pogo pins from the BOM I spent about 2 hours searching aliexpress with little to no luck. I am also having a hard time figuring out what the other two parts that are missing are. Are you able to provide any insight into this?

KRALYN, 2022-05-21:
it is the magnetic connector 6pin; I think it is on line 21 in the current
revision. Also, are you sure you are going to order everything before I
upload the build series and instructions?

KRALYN, 2022-05-21:
Also, DO NOT SMT Solder the connector! De select them. You are going to
solder wires to the pads.

KRALYN, 2022-05-21:
I made BOM more clear in this regard.

closed 2022-05-21 by KRALYN

jmsether, 2022-05-21:
When you say connector, are you talking about U7? And I was planning on getting the board made and things milled and cut from the file. I can handle some of the milling and all the 3d printing.