KRALYN/PositronV3 #5: Open Discussions Option For Community Collaboration

jeffersonbenson, 2022-05-23:
First off, this is a fantastic project that is by far one of the most fascinating designs I've seen in a long time. Can't wait to see your build guide and get my own on my desk.

I'm especially excited for the community that seems to be building around the design. One thing that might be helpful is a space for individuals to talk together on where to source parts or perhaps organize group-buys of the machined components? It would be good to centralize all the discussion in a space that's not the Issues tab. 

Not sure if you have plans to launch a Discord or anything more formal, but I've had good experience with the GitHub Discussions tab, which should be easy to enable from the repository 'Settings' page. 

Again, fantastic project and I hope this suggestion helps 😁

itsthejoker, 2022-05-23:
Someone created a subreddit here: https://www.reddit.com/r/Positron3DPrinter/

closed 2022-05-24 by KRALYN

john5634, 2022-06-10:
Also very excited for a **build** **guide**!!!!!!!!!!!

KRALYN, 2022-10-11:
> I just made a Discord Channel for this purpose also:
> https://discord.gg/CZ5Z42CZ
>
> The invite link seems to be expired. I would like to join too.

New Discord Link: https://discord.gg/SsRZgfud