KRALYN/PositronV3 #11: ZIP is not ideal in git repositories

ascheucher, 2022-05-31:
Hi KRALYN!

That's an awesome project! I am really tempted to rebuild the printer, but let's see whether I find time for that.

Just something I noticed, during checking your repo: ZIP files are not encouraged in git repositories.
You will lose the possibility to track changes and also AFAIK, git compresses everything anyway on transfer and storage.
It's just only drawbacks checking in zipped files.

Thanks for sharing your project!

ascheucher, 2022-06-01:
Never used git lfs, but this sounds like the proper way to do it.

luminous-flux, 2022-06-02:
It works well for CAD - it just tells git to treat the files as pure binary
blobs so it doesnt try to diff them as text with previous versions.

KRALYN, 2022-07-22:
I think this application of Zip is fine since it is an CAD assembly.

closed 2022-07-22 by KRALYN