KRALYN/PositronV3 #12: Need clarification

antordey51, 2022-06-04:
This is in regards to Raw materials/Stock parts available in github repository.

In the parts sourcing excel sheet - In row 109 :

Aluinum Block 6061 MISC macheing 40x40x70 bar1

https://amzn.to/3wI5uVH

The size mentioned in the sheet is 40×40×70 (I assume in mm) but in the provided Amazon link- the size of the AL bar is 50mm×50mm×150mm.

Well, I need a clarification regarding this size issue. Does or doesn't the size matters? If I order from that Amazon link- the size will be bigger than required according to the sheet. So do I have to cut it later to match the size? Honestly, instead of doing that myself, I will better order a customized one if size really matters.

antordey51, 2022-06-04:
@KRALYN

cyberhuman, 2022-06-05:
I think it just needs to be 40x40x70mm or bigger.

KRALYN, 2022-07-22:
The stock sizes are just recommendations, I'll make this clear in the BOM

closed 2022-07-24 by antordey51