KRALYN/PositronV3 #17: Found the 1030 on AliExpress

18Markus1984, 2022-06-16:
[image]

I found the 1030 extrusion on [AliExpress ](https://de.aliexpress.com/item/1005001957312144.html?spm=a2g0o.order_list.0.0.21ef5c5fu1CdhL&gatewayAdapt=glo2deu) maybe you can add the link to the BOM. The size values are at some points a little different, but shouldn't be bothering the build.

KRALYN, 2022-07-22:
Very nice! I think this piece will work as well. I'll add it into the BOM

closed 2022-07-22 by KRALYN

antordey51, 2022-07-24:
@18Markus1984 isn't the one from misumi cheap around 3 USD? I checked the Aliexpress link and for 250mm- it's around 9 USD:(

18Markus1984, 2022-07-24:
> @18Markus1984 isn't the one from misumi cheap around 3 USD? I checked the Aliexpress link and for 250mm- it's around 9 USD:(

Yes but not shippeable to Europe