KRALYN/PositronV3 #37: Why is the hotend limited to 250C? Is it PTFE-lined?

kozross, 2023-02-19:
The title says it all - it seems kind of odd to have a limit like this for any other reason. If the hotend is indeed PTFE-lined, it seems like a very odd choice, since it means you can't print anything other than PLA, really.

metaxis, 2023-02-19:
petg, pctg, asa, abs, tpu, hips, sbs, pva all print well at 250 C and below.  Where did you hear you can only print PLA below 250C?

kozross, 2023-02-19:
@metaxis - ASA needs 260 in practice, and is in fact the recommended temperature by at least one manufacturer. I literally had layer delamination and cracking at 250 with Prusament ASA. PETG can print below that, but you need to go above 250C to get any decent flow - again, speaking from experience. I haven't printed PCTG, ABS, TPU or HIPS, but from a casual glance, many manufacturers of such filaments suggest temps _well_ above 250.

Regardless, my question stands.

nomadsgalaxy, 2025-01-07:
This is super late - hello from 2 years later; Yes, it's because the PTFE tube presses against the heatbreak, and we don't want people to risk getting toxic fumes inhaled. The new 3.2 hotend is easily able to hit 300c, but the PTFE is the dangerous part.

See the latest version of the project here; https://github.com/Positron3D/Positron

closed 2025-01-07 by nomadsgalaxy